MangoAutomation/modbus4j #5: how to use TestSerialWrapper?

robbStarkTFG4, 2016-02-11:
can you add an example showing implementation of interface SerialPortWrapper?.

terrypacker, 2016-02-11:
Here is one that we use inside of Mango Automation.  

You will also need to refer to the abstract class  [SerialPortWrapper](https://github.com/infiniteautomation/ma-core-public/blob/2.7.x/Core/src/com/infiniteautomation/mango/io/serial/SerialPortProxy.java) and one of its concrete implementations [JsscSerialPortProxy](https://github.com/infiniteautomation/ma-core-public/blob/2.7.x/Core/src/com/infiniteautomation/mango/io/serial/JsscSerialPortProxy.java)

`
/**
 * Copyright (C) 2015 Infinite Automation Software. All rights reserved.
 * @author Terry Packer
 */
package com.infiniteautomation.mango

import java.io.InputStream;
import java.io.OutputStream;

import com.infiniteautomation.mango.io.serial.SerialPortProxy;
import com.serotonin.m2m2.Common;
import com.serotonin.modbus4j.serial.SerialPortWrapper;

/**
 * @author Terry Packer
 *
 */
public class MangoSerialPortWrapper implements SerialPortWrapper{
	
	private String ownerName;
	private String commPortId;
    private int baudRate;
    private int flowControlIn;
    private int flowControlOut;
    private int dataBits;
    private int stopBits;
    private int parity;
    
    private SerialPortProxy proxy;
    
	/**
	 * @param ownerName
	 * @param commPortId
	 * @param baudRate
	 * @param flowControlIn
	 * @param flowControlOut
	 * @param dataBits
	 * @param stopBits
	 * @param parity
	 */
	public MangoSerialPortWrapper(String ownerName, String commPortId,
			int baudRate, int flowControlIn, int flowControlOut, int dataBits,
			int stopBits, int parity) {
		super();
		this.ownerName = ownerName;
		this.commPortId = commPortId;
		this.baudRate = baudRate;
		this.flowControlIn = flowControlIn;
		this.flowControlOut = flowControlOut;
		this.dataBits = dataBits;
		this.stopBits = stopBits;
		this.parity = parity;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#close()
	 */
	@Override
	public void close() throws Exception {
		Common.serialPortManager.close(proxy);
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#open()
	 */
	@Override
	public void open() throws Exception {
		this.proxy = Common.serialPortManager.open(ownerName, commPortId, baudRate, flowControlIn, flowControlOut, dataBits, stopBits, parity);
		
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getInputStream()
	 */
	@Override
	public InputStream getInputStream() {
		return this.proxy.getInputStream();
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getOutputStream()
	 */
	@Override
	public OutputStream getOutputStream() {
		return this.proxy.getOutputStream();
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getBaudRate()
	 */
	@Override
	public int getBaudRate() {
		return this.baudRate;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getDataBits()
	 */
	@Override
	public int getDataBits() {
		return this.dataBits;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getStopBits()
	 */
	@Override
	public int getStopBits() {
		return this.stopBits;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getParity()
	 */
	@Override
	public int getParity() {
		return this.parity;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getFlowControlIn()
	 */
	@Override
	public int getFlowControlIn() {
		return this.flowControlIn;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getFlowControlOut()
	 */
	@Override
	public int getFlowControlOut() {
		return this.flowControlOut;
	}

}
`

djdenv, 2016-02-11:
Unfortunately, there is little documentation on this, but here's another one I created using JSSC as the underlying Port implementation. 

```
import com.serotonin.modbus4j.serial.SerialPortWrapper;
import jssc.SerialPort;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

import java.io.InputStream;
import java.io.OutputStream;
import java.util.ArrayList;
import java.util.List;

/**
 * Created by Domenic on 12/22/2015.
 */
public class SerialPortWrapperImpl implements SerialPortWrapper {
    private static final Logger LOGGER = LoggerFactory.getLogger(SerialPortWrapperImpl.class);
    private final SerialPort port;
    private List<PortConnectionListener> listeners = new ArrayList<>();

    public void addPortConnectionListener(PortConnectionListener listener) {
        listeners.add(listener);
    }

    public SerialPortWrapperImpl(String portName) {
        port = new SerialPort(portName);
    }

    @Override
    public void close() throws Exception {
        port.closePort();
        listeners.forEach(PortConnectionListener::closed);
        LOGGER.debug("serial port " + port.getPortName() + " is closed");
    }

    @Override
    public void open() throws Exception {
        port.openPort();
        port.setParams(this.getBaudRate(), this.getDataBits(), this.getStopBits(), this.getParity());
        port.setFlowControlMode(this.getFlowControlIn() | this.getFlowControlOut());

        listeners.forEach(PortConnectionListener::opened);
        LOGGER.debug("serial port " + port.getPortName() + " is open");
    }

    @Override
    public InputStream getInputStream() {
        return new SerialInputStream(port);
    }

    @Override
    public OutputStream getOutputStream() {
        return new SerialOutputStream(port);
    }

    @Override
    public int getBaudRate() {
        return SerialPort.BAUDRATE_9600;
    }

    @Override
    public int getFlowControlIn() {
        return SerialPort.FLOWCONTROL_NONE;
    }

    @Override
    public int getFlowControlOut() {
        return SerialPort.FLOWCONTROL_NONE;
    }

    @Override
    public int getDataBits() {
        return SerialPort.DATABITS_8;
    }

    @Override
    public int getStopBits() {
        return SerialPort.STOPBITS_1;
    }

    @Override
    public int getParity() {
        return SerialPort.PARITY_NONE;
    }
}

```

cevatbostancioglu, 2016-12-19:
hi. i am using rs232 interface with rs485 converter ic and i need to down rts pin. do you know how i can do this with jSerialComm library and why you wrote both getflowcontrol override functions flowcontrol_none  ?

SharPeiDog, 2017-07-13:
本人小白一个 以下只是我自己的做法，有什么不足希望大家可以指出。
首先，你需要下载一个额外的支持Java串口通信操作的jar包。
地址：
[http://files.cnblogs.com/files/Dreamer-1/mfz-rxtx-2.2-20081207-win-x86.zip](http://files.cnblogs.com/files/Dreamer-1/mfz-rxtx-2.2-20081207-win-x86.zip )（32位）

[http://files.cnblogs.com/files/Dreamer-1/mfz-rxtx-2.2-20081207-win-x64.zip](http://files.cnblogs.com/files/Dreamer-1/mfz-rxtx-2.2-20081207-win-x64.zip) （64位）

.dll文件放在jdk路径下（xx\jre\bin）
.jar 文件导入项目 
具体操作参考：
[http://www.cnblogs.com/Dreamer-1/p/5523046.html](http://www.cnblogs.com/Dreamer-1/p/5523046.html)

接下来是我的 SerialPortWrapper 实现类

`
package com.serotonin.modbus4j.test.mytest;

import java.io.IOException;
import java.io.InputStream;
import java.io.OutputStream;

import com.serotonin.modbus4j.serial.SerialPortWrapper;

import gnu.io.CommPort;
import gnu.io.CommPortIdentifier;
import gnu.io.SerialPort;
import gnu.io.UnsupportedCommOperationException;

/**
 * 
 * This class is not finished
 * 
 * @author Terry Packer
 *
 */
public class SerialPortWrapperImpl2 implements SerialPortWrapper{
	private SerialPort serialPort ;
	private String commPortId;
    private int baudRate;
    private int flowControlIn;
    private int flowControlOut;
    private int dataBits;
    private int stopBits;
    private int parity;
    public SerialPortWrapperImpl2(){
    	super();
    }
	public SerialPortWrapperImpl2(String commPortId, int baudRate, int flowControlIn,
			int flowControlOut, int dataBits, int stopBits, int parity){
		this.commPortId = commPortId;
        this.baudRate = baudRate;
        this.flowControlIn = flowControlIn;
        this.flowControlOut = flowControlOut;
        this.dataBits = dataBits;
        this.stopBits = stopBits;
        this.parity = parity;
	}
	

	
	public String getCommPortId() {
		return commPortId;
	}
	public void setCommPortId(String commPortId) {
		this.commPortId = commPortId;
	}
	public void setBaudRate(int baudRate) {
		this.baudRate = baudRate;
	}
	public void setFlowControlIn(int flowControlIn) {
		this.flowControlIn = flowControlIn;
	}
	public void setFlowControlOut(int flowControlOut) {
		this.flowControlOut = flowControlOut;
	}
	public void setDataBits(int dataBits) {
		this.dataBits = dataBits;
	}
	public void setStopBits(int stopBits) {
		this.stopBits = stopBits;
	}
	public void setParity(int parity) {
		this.parity = parity;
	}
	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#close()
	 */
	@Override
	public void close() throws Exception {
		// TODO Auto-generated method stub
		  if (serialPort != null) {
	            serialPort.close();
	            serialPort = null;
	        }
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#open()
	 */
	@Override
	public void open() throws Exception {
		 System.out.println(commPortId);
		// TODO Auto-generated method stub
		 CommPortIdentifier portIdentifier = CommPortIdentifier.getPortIdentifier(commPortId);

         //打开端口，并给端口名字和一个timeout（打开操作的超时时间）
		 System.out.println(commPortId);
         CommPort commPort = portIdentifier.open(commPortId, 2000);

         //判断是不是串口
         if (commPort instanceof SerialPort) {
             
             serialPort = (SerialPort) commPort;
             
             try {                        
                 //设置一下串口的波特率等参数
                // serialPort.setSerialPortParams(baudRate, SerialPort.DATABITS_8, SerialPort.STOPBITS_1, SerialPort.PARITY_NONE);                              
             } catch (Exception e) {  
             }
             
             //System.out.println("Open " + portName + " sucessfully !");
            // return serialPort;
         
         }        
         else {
             //不是串口
        	 System.out.println("不是串口");
         }
		
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getInputStream()
	 */
	@Override
	public InputStream getInputStream() {
		InputStream in = null;
		try {
			in = serialPort.getInputStream();
		} catch (IOException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
		
		return in;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getOutputStream()
	 */
	@Override
	public OutputStream getOutputStream() {
		OutputStream out = null;
		try {
			out = serialPort.getOutputStream();
		} catch (IOException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
		
		return out;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getBaudRate()
	 */
	@Override
	public int getBaudRate() {
		// TODO Auto-generated method stub
		return 0;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getStopBits()
	 */
	@Override
	public int getStopBits() {
		// TODO Auto-generated method stub
		return 0;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getParity()
	 */
	@Override
	public int getParity() {
		// TODO Auto-generated method stub
		return 0;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getFlowControlIn()
	 */
	@Override
	public int getFlowControlIn() {
		// TODO Auto-generated method stub
		return 0;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getFlowControlOut()
	 */
	@Override
	public int getFlowControlOut() {
		// TODO Auto-generated method stub
		return 0;
	}

	/* (non-Javadoc)
	 * @see com.serotonin.modbus4j.serial.SerialPortWrapper#getDataBits()
	 */
	@Override
	public int getDataBits() {
		// TODO Auto-generated method stub
		return 0;
	}

}

`
BY：一只等待星期五的沙皮狗