MangoAutomation/modbus4j #6: Setting up modbus device without automatic RX/TX handling

natami, 2016-04-25:
Is there a way to make Modbus4J support enablement of RX/TX handling for devices without automatic detection?

terrypacker, 2016-04-26:
I'm not sure I understand exactly what you want to do.  Can you explain this a little better?

Are you talking about CTS,RTS signals?

Thanks!

natami, 2016-04-26:
Hi, Terry!

I've in the past used this USB to RS485 converter with success - which has
automatic receive/transmit switching:
http://www.hjelmslund.dk/Files/Product%20sheet%20-%20USB485-STISO-STIXL.pdf

Now i'm trying the same with this device (coupled with a Raspberry PI):
https://www.sparkfun.com/products/13706
http://cdn.sparkfun.com/datasheets/Dev/Arduino/Shields/max481csa.pdf
http://learn.linksprite.com/raspberry-pi/shield/user-tips-of-rs485-shield-for-rpi/

This new converter doesn't appear to support automatic receive/transmit
switching - which i'm trying to determine if this can be handled by the
Modbus4J implementation - and how?

Hope this clears things up a bit ;)

// Kindly regards

terrypacker, 2016-04-26:
I couldn't see how that device lets you put it into Rx or Tx mode, but assuming there is a way here is how I would attempt this:

The overall plan is to extend an output stream that will switch into Tx mode for writes and at all other times leave the transceiver in Rx mode.

1.  Implement an Output Stream that wraps your current serial port output stream.  This stream should switch the 485 device into and out of transmit mode whenever a write() method is called.  From memory all the Modbus4J calls to the output stream use the write(byte[]...) methods.  So you can probably avoid switching into and out of Rx/Tx modes when sending single bytes and only do this for the full message writes.

2.  Implement a concrete class using SerialPortWrapper that returns your wrapped Output Stream from getOutputStream().  Use this class as the wrapper that is passed into Modbus4J.

3.  If necessary you could implement an InputStream that ensures it is in Rx mode while trying to do reads although this might not be necessary.

Note you may need something like [http://pi4j.com/](http://pi4j.com/) to gain access to toggle the pin on that device to change the Rx/Tx mode.