MangoAutomation/modbus4j #8: Serial port connection

Kosmachevskiy, 2016-12-08:
Hi!

I have some trouble with ModbusSerialMaster. I got a NPE during reading registers.

 
My code:

`

        SerialParameters parameters = new SerialParameters();

        parameters.setPortName("/dev/ttyUSB1");
        parameters.setBaudRate(9600);
        parameters.setDatabits(8);
        parameters.setStopbits(1);
        parameters.setParity("none");
        parameters.setEncoding("rtu");
        parameters.setEcho(false);

        ModbusSerialMaster master = new ModbusSerialMaster(parameters);

        master.connect();
        master.readMultipleRegisters(1, 0, 100);
        master.disconnect();
`
And stack:

`Exception in thread "main" java.lang.NullPointerException
	at com.ghgande.j2mod.modbus.net.SerialConnection.setComPortTimeouts(SerialConnection.java:221)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.setTimeout(ModbusSerialTransport.java:142)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.open(ModbusSerialTransport.java:131)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.writeMessage(ModbusSerialTransport.java:73)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransaction.execute(ModbusSerialTransaction.java:143)
	at com.ghgande.j2mod.modbus.facade.AbstractModbusMaster.readMultipleRegisters(AbstractModbusMaster.java:236)`

Kosmachevskiy, 2016-12-08:
Sorry, it should be a message to other library.

closed 2016-12-08 by Kosmachevskiy

cevatbostancioglu, 2016-12-08:
Hi , you have to download modbus4j source and reference your project if you want to solve serial library problems in linux.
Then , 
- first you are using j2mod library. remove it. 
- you can use fazecastj SerialComm, i tested yesterday both linux and windows are fine. i don't know how to behave at mac os;
- modbus4j source code is little bit different from jar. actually jar files are little bit old, they made a lot of nice stuff but if you use jar files , you can not access it. 
- for rtu/ascii protocols, they made serialwrapper interface. so you can use whatever you want as a serial port library. 

i am kind of busy right now , so i can't write too much. you have to figure out , it's easy.

import java.io.InputStream;
import java.io.OutputStream;
import java.io.File;
import java.io.IOException;

import com.fazecast.jSerialComm.SerialPort;
------

public static SerialPort serialLocalPort;

rtuSerialPort = SerialPort.getCommPort("/dev/tty0");

			serialLocalPort.setBaudRate(115200);
			serialLocalPort.setParity( 0);
			serialLocalPort.setNumDataBits(8);
			serialLocalPort.setNumStopBits(1);

			SerialPortWrapper spw = new SerialPortWrapper()
			{
				@Override
				public void open() throws Exception
				{
					// TODO Auto-generated method stub
					if (serialLocalPort.openPort())
					{
						
					}
				}

				@Override
				public int getStopBits()
				{
					return serialLocalPort.getNumStopBits();
				}

				@Override
				public int getParity()
				{
					return serialLocalPort.getParity();
				}

				@Override
				public OutputStream getOutputStream()
				{
					return serialLocalPort.getOutputStream();
				}

				@Override
				public InputStream getInputStream()
				{
					return serialLocalPort.getInputStream();
				}

				@Override
				public int getFlowControlOut()
				{
					return 0;
				}

				@Override
				public int getFlowControlIn()
				{
					return 0;
				}

				@Override
				public int getDataBits()
				{
					return serialLocalPort.getNumDataBits();
				}

				@Override
				public int getBaudRate()
				{
					return serialLocalPort.getBaudRate();
				}

				@Override
				public void close() throws Exception
				{
					if (serialLocalPort.closePort())
					{
						
					}
				}
			};

			master = factory.createRtuMaster(spw);
		}

Kosmachevskiy, 2016-12-08:
@cevatbostancioglu Thanks a lot!
I am going to use this library.

Kosmachevskiy, 2016-12-08:
@cevatbostancioglu I tried to run your code - it is works. Thanks a lot!

cevatbostancioglu, 2016-12-08:
your welcome.