MangoAutomation/modbus4j #13: How to use this with Gradle?

dmitrykolesnikovich, 2017-08-03:
How about Gradle?

`compile 'com.infiniteautomation:modbus4j:3.0.3'`

dont work for me.

xcerx, 2017-10-02:
I think you should add maven repository for modbus4j by yourself. follow below
```
repositories {
    maven {
         url "https://maven.mangoautomation.net/repository/ias-release/"
    }
}
```

Ronjea, 2017-11-16:
Does did work for you @dmitrykolesnikovich ???

otto-dev, 2019-02-28:
Found this via google, and since it's still open, the following works for sure:

```
repositories {
    mavenCentral()
    maven {
        url 'https://maven.mangoautomation.net/repository/ias-snapshot/'
    }
    maven {
        url 'https://maven.mangoautomation.net/repository/ias-release/'
    }
}

dependencies {
    compile 'com.infiniteautomation:modbus4j:3.0.3'
}
```

So, yes, works, I can import and use the module.

I haven't tried without the `snapshot` repository, it might also work without it.

terrypacker, 2019-02-28:
The repository you need to include depends on what version you are trying to use in your project:

For any non-release version of Modbus4J you will need to include the `snapshot` repo.  So you won't need it for

```
dependencies {
    compile 'com.infiniteautomation:modbus4j:3.0.3'
}
```

but you would need it for

```
dependencies {
    compile 'com.infiniteautomation:modbus4j:3.0.3-SNAPSHOT'
}
```

dmitrykolesnikovich, 2019-02-28:
Sorry for delay, by adding repo as @otto-dev said I fixed an issue.

closed 2019-02-28 by dmitrykolesnikovich