MangoAutomation/modbus4j #19: Modbus RTU over TCP

deinok, 2018-03-07:
Hi, is there any support for this kind of conection?

terrypacker, 2018-03-07:
There is no support 'out of the box', but I would suspect it would work with some additional coding.  If you were to provide a SerialPortWrapper that uses a TCP/IP Socket you should be able to create an RtuSlave or RtuMaster.  We have done something like this is Mango Automation which we have called a SerialSocketBridge and SerialServerSocketBridge.  I haven't personally tested Mango with a ModbusRTU data source using one of these virtual serial ports so I can't make any guarantees.  The code is open source so feel free to poke around and see what you can come up with.

But that would be a good place to start.

deinok, 2018-03-07:
Good to know, are PR accepted for this?

terrypacker, 2018-03-07:
Absolutely.

deinok, 2018-03-07:
Will send it ASAP

terrypacker, 2018-03-07:
Here is the source for our implementation:
[
https://github.com/infiniteautomation/ma-core-public/blob/main/Core/src/com/infiniteautomation/mango/io/serial/virtual/SerialServerSocketBridge.java](
https://github.com/infiniteautomation/ma-core-public/blob/main/Core/src/com/infiniteautomation/mango/io/serial/virtual/SerialServerSocketBridge.java)

closed 2019-07-10 by terrypacker

minghu, 2019-07-12:
i  have checked ,rtu over tcp  Communication message actually  is same as  tcpmaster use   encapsulated，so we can use the tcpmaster  with encapsulated option,and EncapMessage is same as  RtuMessage,just communiction channel different.of course，rtu ove tcp is one more choice.

terrypacker, 2019-07-12:
Thanks, just worked that out.  I'll remove the class in the corresponding PR from the codebase then.

PR #20 (2019-07-10): Rtu Over TCP
Closes #19 

Seems working well with a USR-WIFI232-602 that exposes everything in serial port (Schneider M221) via a TCP server. (RTU over TCP)

Line 66 it is necesary. 
But if there is a better wait to detect when te TCP PDU is received would be better.
commit 9eaba7e7 (2019-07-10): Fixes #19 RTU over TCP thanks @deinok