MangoAutomation/modbus4j #30: "Illegal function" response when read/write coils

yanxin9210, 2019-08-16:
It returns "Illegal function" when I read or write coils.
Version: 3.0.5
Code: 
```
    public static void main(String[] args) {
        IpParameters ipParameters = new IpParameters();
        ipParameters.setHost("192.168.31.200");
        ipParameters.setPort(502);

        ModbusFactory modbusFactory = new ModbusFactory();
        ModbusMaster master = modbusFactory.createTcpMaster(ipParameters, false);
        int slaveId = 1;
        try {
            master.init();
            readCoilTest(master, slaveId, 0, 1);
        } catch (ModbusInitException e) {
            e.printStackTrace();
        }
    }

    public static void readCoilTest(ModbusMaster master, int slaveId, int start, int len) {
        try {
            ReadCoilsRequest request = new ReadCoilsRequest(slaveId, start, len);
            ReadCoilsResponse response = (ReadCoilsResponse) master.send(request);

            if (response.isException()) {
                System.out.println("Exception response: message=" + response.getExceptionMessage());
            } else {
                System.out.println(Arrays.toString(response.getBooleanData()));
            }
        } catch (ModbusTransportException e) {
            e.printStackTrace();
        }
    }
```

Logs:
```
10:58:15.121 [main] DEBUG com.serotonin.modbus4j.ip.tcp.TcpMaster - Encap Request: 00 01 00 00 00 06 01 01 00 00 00 01 
10:58:15.137 [main] DEBUG com.serotonin.modbus4j.ip.tcp.TcpMaster - Sending on port: 502
10:58:15.219 [main] DEBUG com.serotonin.modbus4j.ip.tcp.TcpMaster - Response: 00 01 00 00 00 03 01 81 01 
Exception response: message=Illegal function
```

minghu, 2019-08-18:
see your  log,00 01 00 00 00 03 01 81 01 , 81 is error code,normally,it will  return   function  code，so the 81 is Illegal function code ,please  use the modscan32 check you slave device.
      which you company is?

Owisky, 2021-05-31:
your slave modbus server may not support some functions.

kay07, 2023-06-14:
The reason is that you have no create Mbslave in modbus slave