MangoAutomation/modbus4j #32: How to modify the default timeout when using the Modbus RTU Master?

duzean, 2019-09-19:
I have a remote equipement to be communicated with the Modbus RTU protocol. I use the Netty framework deal with the TCP/IP layer, writting a wrapper of SerialPort by the Netty, I did it. But when I test my case, I found modbus4j sometimes send a request twice, I checked the source code, found the timeout is set to 500 default.Sometimes 500ms is too short for the TCP/IP communication. So I set the RtuMaster timeout to 3000ms, unfortunately, it does't work. I found the timeout parameter is useless for the Modbus RTU Master, the valid one in the MessageControl class. So how to change the default timeout value in the MessageControl? I can‘t’ find a way achieve it.

terrypacker, 2019-09-20:
I haven't tested what you are saying, but from looking at the code you need to set the timeout on the `RtuMaster` class **before** you call `RtuMaster.init()`.  `ModbusMaster.timeout` is used when during `RtuMaster.init()` when it calls `getMessageControl()` to setup the message control.  Setting it after that will have no effect as the `MessageControl` connection is already created.

duzean, 2019-09-20:
Thanks for your reply. You are right. I put the setting of timeout after init. Thank you very much!

closed 2019-09-20 by terrypacker