MangoAutomation/modbus4j #35: Floating Point Number is Not Accurate in modbusTcp

JunshiJia, 2020-01-09:
Hello, recently I use lots of the read and write FOUR_BYTE_FLOAT function(batchRead) in your MODBUS TCP module, I found the number is not accurate(maybe my own problem).  More specific, the float numbers generally have  an error of 0.5%, which is not normal for floating point numbers.  

I am guessing, is it possible that the 3rd and 4th bytes are flipped that causes this problem?

Does anyone else has this problem?

JunshiJia, 2020-01-09:
So I did more test. I use your module to Write 30000 float point number on a slave address for 4bytes float, and use another java module(called easy modbus) to read this addres, i got 299584.0.  This should be a bug, right? 

Wrote 3.0E7, read 2.992128E7.

minghu, 2020-01-10:
@JunshiJia ，this is not  modbus4j bug, may be occured  when you use the value,you can use the java  BigDecimal instead of the double or flout,it have the accuracy problem.

JunshiJia, 2020-01-10:
@minghu Actually I think you are right.  I did more tests found my bug. I am truly an amateur got to admit that.

closed 2020-01-10 by terrypacker