MangoAutomation/modbus4j #52: run modbus4j error

zhouhangzooo, 2022-03-30:
I’m importing modbus 4j-2.0.2. Jar and JSSC-2.8.0. Jar, error running project as follows:

java.lang.NoSuchMethodError: com.serotonin.io.serial.SerialUtils.openSerialPort
        at com.serotonin.modbus4j.serial.SerialMaster.init(SerialMaster.java:58)
        at com.serotonin.modbus4j.serial.rtu.RtuMaster.init(RtuMaster.java:45)

 I need a solution,thank your!

zhouhangzooo, 2022-03-30:
I found the missing RXTXPort and SerialPort in the MODBUS 4J project, which you referenced, and the code is as follows:
import gnu.io.RXTXPort;
import gnu.io.SerialPort;

terrypacker, 2022-03-30:
That is a very old version of modbus4J which likely depends on RxTx Java not JSSC for serial port support.

closed 2022-03-30 by terrypacker

zhouhangzooo, 2022-03-31:
> That is a very old version of modbus4J which likely depends on RxTx Java not JSSC for serial port support.

What are the latest jars you use in Modbus4j? What is the exact version?

zhouhangzooo, 2022-03-31:
> > That is a very old version of modbus4J which likely depends on RxTx Java not JSSC for serial port support.
> 
> What are the latest jars you use in Modbus4j? What is the exact version?
I used mobus4j. Jar and RXTXCOMM. Jar, there was still an error.
java.lang.UnsatisfiedLinkError: Native method not found: jssc.SerialNativeInterface.openPort:(Ljava/lang/String;Z)J
        at jssc.SerialNativeInterface.openPort(Native Method)
        at jssc.SerialPort.openPort(SerialPort.java:158)
        at com.example.socketapp.modbus4j.util.TestSerialPortWrapper.open(TestSerialPortWrapper.java:52)
        at com.example.socketapp.modbus4j.serial.SerialMaster.openConnection(SerialMaster.java:100)
        at com.example.socketapp.modbus4j.serial.rtu.RtuMaster.openConnection(RtuMaster.java:85)
        at com.example.socketapp.modbus4j.serial.rtu.RtuMaster.init(RtuMaster.java:74)

terrypacker, 2022-04-01:
https://github.com/MangoAutomation/modbus4j/releases/tag/v3.1.0

Use Maven to manage your dependencies.  See the README.md file for where the builds are located.