MangoAutomation/modbus4j #64: 请问各位大佬，No recipient was found waiting for response for key

zj555, 2023-05-10:
不定时出现如下异常，这有什么影响吗？该怎么解决呢？
com.serotonin.modbus4j.sero.messaging.WaitingRoomException: No recipient was found waiting for response for key com.serotonin.modbus4j.ip.xa.XaWaitingRoomKeyFactory$XaWaitingRoomKey@864a
        at com.serotonin.modbus4j.sero.messaging.WaitingRoom.response(WaitingRoom.java:100)
        at com.serotonin.modbus4j.sero.messaging.MessageControl.data(MessageControl.java:198)
        at com.serotonin.modbus4j.sero.messaging.InputStreamListener.run(InputStreamListener.java:76)
        at java.lang.Thread.run(Thread.java:748)

kay07, 2023-06-12:
我也出现了同样的问题，网上说只能master主动向slave发起请求，不能slave主动发起，这个说法本身没问题但是感觉没说到点子上。我出现这个问题是因为建立了一个连接对象，然后持续用这个连接对象，如果每次请求都用一个新建的连接对象就不会有这个问题了，所以是不是这个连接对象只能用一次就销毁了，所以出现这个问题的？

kay07, 2023-06-12:
我也出现这个问题了。我的是因为设置ModbusMaster的setTimeout太小了导致的

moyaoo, 2023-10-30:
slave id不能为0，我这用modbus slave测试时，slave id设置为0时始终报错，1就可以

kay07, 2025-07-02:
超时时间设置大点也不行吗

开心
***@***.***

fdj0926, 2025-07-02:
> 超时时间设置大点也不行吗
> 
> 
> 
> 
> 开心
> ***@***.***

也不行呢。。。

fdj0926, 2025-07-02:
> 超时时间设置大点也不行吗
> 
> 
> 
> 
> 开心
> ***@***.***

也不行呢，我是这样调用的：ModbusUtil.readInputRegister("10.114.38.1", 503, 1, 970, DataType.TWO_BYTE_INT_UNSIGNED);